bambulab/BambuStudio #2365: Support object cancellation when using multiple materials (or at least when using dedicated support material).

ADKMechETECH, 2023-08-23:
**Is your feature request related to a problem? Please describe.**
Object cancellation does not work when multiple materials (even support material) are used.

**Describe the solution you'd like**
At the very least make object cancellation work with support material.  Ideally with multiple materials since the whole point of having an AMS that can automatically switch materials is to print with more than one material in a single print.

**Describe alternatives you've considered**
Printing without supports, or print with supports with the same material as the parts.  This is not ideal as some parts *REQUIRE* supports, and printing supports with the same material doesn't give acceptable results in all cases.

**Additional context**
None.

SaltWei, 2023-08-23:
Please update to 1.7.4 bambu studio

closed 2023-08-23 by SaltWei

SaltWei, 2023-08-24:
Because of human resources limitation, object cancellation can only be available on X1C. P1P will be later