bambulab/BambuStudio #2367: Building a Windows environment will always report errors dep_OpenEXR-prefix and dep_wxWidgets-prefix

ZZQiang-An, 2023-08-23:
Can anyone help me look at this???
[image]
[image]

lanewei120, 2023-08-28:
oops, just missed this issue before

do you still have this issue?

it seems the patch stage of wxwidget failed for the second time
you can remove dep_wxWidgets folder and try again

sorry for the late reply

ZZQiang-An, 2023-09-08:
[image]
What about this error?
@lanewei120

lanewei120, 2023-09-08:
> [image] What about this error? @lanewei120

please send the full log of building error

ZZQiang-An, 2023-09-08:
> > [image]这个错误呢？
> 
> 请发送完整的建筑错误日志

[build BambuStudio log.txt](https://github.com/bambulab/BambuStudio/files/12556797/build.BambuStudio.log.txt)
[image]
[image]
What causes this error to show up when opening? Can you tell me?

lanewei120, 2023-09-08:
the building error consist of two reasons:
1. wxWidgets patch failed
to fix it, you need to rm C:\Slicer\Bambu\BambuStudio\deps\build\dep_wxWidgets-prefix, and rebuild again
2. networking error
error : RPC failed; curl 18 HTTP/2 stream 5 was reset
as lots of codes are on github, you need to use a better network connection

lanewei120, 2023-09-08:
> hat causes this error to show up when opening? Can you tell m

@bambu123 could you check the edge related issue

bambu123, 2023-09-08:
> > > [image]这个错误呢？
> > 
> > 
> > 请发送完整的建筑错误日志
> 
> [build BambuStudio log.txt](https://github.com/bambulab/BambuStudio/files/12556797/build.BambuStudio.log.txt) [image] [image] What causes this error to show up when opening? Can you tell me?

Install a public version of BambuStudio first.

lanewei120, 2023-12-26:
we have found a building error of OpenEXR and solved at https://github.com/bambulab/BambuStudio/pull/3188

closed 2023-12-26 by lanewei120