bambulab/BambuStudio #2368: Calibration dos not Work with 0.6 nozzle

Champino74, 2023-08-23:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.5.1

### Additional system information

MacBook Pro
3.1 GHz Quad-Core Intel Core i7
Radeon Pro 560 4 GB
Intel HD Graphics 630 1536 MB
16 GB 2133 MHz LPDDR3

### Printer

Bambu Lab X1C

### How to reproduce

1. go to calibration
2. Flow Dynamics
3. change nozzle to 0.6 
4. klick calibrate

### Actual results

Invalid nozzle diameter
[image]

### Expected results

calibration works 

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12416392/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Champino74, 2023-08-23:
OK, thank you

majianjia, 2024-01-16:
@Champino74

dragon-eyes-bbl, 2024-01-16:
@Champino74 hi, 0.6mm nozzle has been supported. Please update Bambu Studio to 1.8.2 or latest to use this function. Thanks

closed 2024-01-16 by dragon-eyes-bbl