bambulab/BambuStudio #2370: [Feature Request] Allow multiple projects open in Bambu Studio

BambuLabDiscord, 2023-08-23:
Many times one wants to work on more than one project, possibly having different project printing on different printers, at the same time. It will be helpful to: 

1) Have multiple projects open, each its own tab. 
2) When print directly from CAD apps, such as Fusion 360, the model should go to the active plate on the active project (the project that's open on screen).
3) Each project tab should have access to all of the printers.
4) This feature should also work on LAN Only mode

werewolfenn, 2023-08-23:
YES!   So far, the best option has been to add a plate for each machine, then try to remember to only save the plate you are currently tweaking.

QingZhangBambu, 2023-08-24:
I will add it to our feature pool, the concerned staff will evaluate it.

GlennovitS, 2024-04-22:
Any news on this one? Just got my second printer, and I find myself doing multiple projects all the time - would be awesome to have them all open with tabs in Bambu Studio! 😄 Will be happy when/if it comes - no-matter when (but hopefully sooner rather than later 😜) Thanks in advance 👍🏼