bambulab/BambuStudio #2372: Crash on Mac when reconnecting to a printer via LAN

DarrenLynch, 2023-08-23:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS Monterey

### Printer

P1S

### How to reproduce

1. Open bambu lab
2. Open a project
3. slice and print
4. After the print starts, close the MacBook
5. Some time later, re-open MacBook
6. Navigate to devices, when your printer (LAN) re-appears, connect
7. Bambu labs crashes
8. When you open it again you are prompted to recover project

### Actual results

Bambu labs crashes

### Expected results

No crash, printer connects and you can see its progress and status as expected

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/12422218/Archive.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

walterwongbbl, 2023-08-28:
Hi.
We will fix this issue

walterwongbbl, 2023-08-28:
@DarrenLynch  
Can you provide a console log. the studio should have experienced a crash
[image]
[image]

DarrenLynch, 2023-08-28:
@walterwongbbl sure:
[bambuCrashReports.zip](https://github.com/bambulab/BambuStudio/files/12453094/bambuCrashReports.zip)

walterwongbbl, 2023-10-13:
Hi. @DarrenLynch 
Please download and use the latest Studio version

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl