bambulab/BambuStudio #2375: Support Interface - First Layer of interface using wrong material

jennyy867, 2023-08-24:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

MacOS Ventura 13.1 

### Additional system information

CPU: 3 GHz 6-Core Intel Core i5
Memory: 8 GB 2667 MHz DDR4
Graphics: Radeon Pro 570X 4 GB

### Printer

BambuLab X1C-Combo

### How to reproduce

1. Using a model that requires support, check on the support function and set your interface to "Support for PLA", leave raft/base set to default or select PLA. 
2. Uncheck "flush into supports". Support Type is Tree Auto, 5.5 branch distance, 3 branch diameter. 2 top interface layers, concentric pattern.
3. Slice the model with a different color PLA from Supports
4. Go to preview and view filament color. Slicing all the way down to the start of the support interface, you'll see a concentric pattern of support in the same PLA color (non support).
5. Switch to Line Type and you can confirm it is a Support line. 

### Actual results

The slicer is placing an "interface layer" made of PLA, not support material, despite the settings. This means the support cannot actually be removed. Compare layer 40, 41 (bad), and 42 in the model.

[image]
[image]

[image]
[image]
[image]
[image]
[image]

### Expected results

Because "support for PLA" is chosen as the interface material, the part touching the model should be printed with support material, not the model's PLA. 

### Project file & Debug log uploads

[BambuBug-InterfaceIssue.3mf.zip](https://github.com/bambulab/BambuStudio/files/12428987/BambuBug-InterfaceIssue.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

tangketan, 2023-09-06:
Thanks for reporting. It's a known issue. We have fixed it in our internal version. Please wait for the next release.

jennyy867, 2023-09-06:
Thank you! Any idea when that might be? I rolled back to an older version but I'd sure like to use the nifty calibration tools in the new one

jennyy867, 2023-09-25:
@tangketan I just checked and this behavior persists in the newest release from 3 days ago (1.7.6.92).  Do you need a new ticket, or did it just not make it into the release?

tangketan, 2023-12-21:
Offical 1.8 has been released and this bug is fixed. Close this issue.

closed 2023-12-21 by tangketan