bambulab/BambuStudio #2376: 'Zoom to mouse location'

megamaniac, 2023-08-24:
So currently in the slicer the zoom function zooms in and out to the centre of the screen. 

This is workable but makes navigation slow - you have to drag the point you want to zoom in on to the what you *think* is the centre of the screen, then zoom a little, make adjustments and zoom again. You can get the job done but it's tedious and slow.

Navigation would be FAR more fluid if instead it mirrored the way this works in a CAD application like Fusion360 where the window zooms to the mouse pointer. This makes it possible to move around the current orientation with a simply zoom out and back in again.

DokuKaefer, 2023-08-26:
Have a look in the settings

[image]

megamaniac, 2023-08-26:
Well dang... 

Well why it isn't default I'll never know. Surely it's preferable.

lanewei120, 2024-06-26:
already included

closed 2024-06-26 by lanewei120