bambulab/BambuStudio #2377: In "Send print job to" page, PLA-Aero is recognize as PLA-Basic. So Can't print with PLA-Aero preset

xuhao1, 2023-08-24:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Printer

X1CC

### How to reproduce

1. Slice 3d models with Bambu PLA Aero material
[image]
In fiament dropbox, the PLA-Aero is PLA-Aero, also on machine screen.
[image]

2. Print

### Actual results

In "send job" page, the PLA-Aero will be recognize as PLA. So I can't print with it.

### Expected results

I want print PLA-Aero with AMS.
I have use FusRock PLA-Aero with AMS, and it is quiet good.

### Project file & Debug log uploads

[Wings.3mf.zip](https://github.com/bambulab/BambuStudio/files/12430556/Wings.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

a1axx, 2023-08-29:
Just to add to this.

I have been printing with Bambu PLA-Aero.

1: the work around is to set go to your filament preset for PLA-Aero and change it to PLA, it will then allow you to print from the AMS

2: as the filament isn't correctly supported, K factor has to be manually selected before printing after a power cycle. It 'sticks' for all other Bambu filaments in the AMS

xuhao1, 2023-09-03:
@a1axx Thanks! It works!

StoneLiBambulab, 2023-10-10:
Hi, 
the filament type in the filament preset of Bambu Studio and the filament type reading from the real filament did not match. we found a way work around it in release 1.8.

Thanks!

Stone Li

SaltWei, 2024-01-18:
[image]
It's OK under latest 1.8 version and latest firmware.
Thanks for reporting issue for us.

closed 2024-01-18 by SaltWei