bambulab/BambuStudio #2380: Toggle between "metric" and "imperial" without going into settings.

shawnishihara, 2023-08-24:
Sometimes is easier to visual the scale of something using a different units. It would be nice to have a toggle in the "scale" tool, or somewhere one can quickly switch between mm/in.

EklDuderino, 2023-09-03:
I 2nd this addition. Prusa slicer has a check box that toggles on the fly between metric and imperial measurements. This would be a great improvement. Please add this feature. Thank you!

EricBKeller, 2024-11-12:
Any information regarding this request? Pretty much a deal breaker for me as to whether I will use Bambu Studio or not. All of my measuring devices, calipers, micrometers, scales, etc. are in Imperial units. I would not consider 3D printing in Metric units.