bambulab/BambuStudio #2381: Open Source the network plugin

andrewhamon, 2023-08-24:
**Is your feature request related to a problem? Please describe.**
In order to sent prints over the network, an additional network plugin must be downloaded and installed (on linux this is saved to `~/.config/BambuStudio/plugins/libbambu_networking.so`).

I use NixOS, and although I am able to get Bambu Studio running with the appimage, I am not able to use the network plugin.

**Describe the solution you'd like**
The source for `libbambu_networking.so` is made available, so that I can build it for the paltforms I care about (NixOS, in my case).

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

ClayShoaf, 2024-04-16:
It seems that this would be a violation of the AGPL, which this software is licensed under. Maybe I am missing something. @bambulab, have you communicated with @prusa3d or @slic3r about this?

Natureshadow, 2024-07-11:
We explicitly chose to buy a BambuLab P1S because it is promoted as being operational using free software.

However, it turned out that we were fooled.

Please fulfill your promises and provide the source code of the network plug-in!

makkarpov, 2024-07-11:
I could say the same about disappointment in Bambulab products after purchase of X1E. This is the first and last Bambulab printer that I ever purchased.

What looks like an open-source software at the first glance, in reality turns out to be a local branch of Apple. You get a locked-up product, but unlike Apple it falls apart right away. When it does (approximately 10 minutes after box is opened), you are left with compiled blobs and encrypted debug logs.

leggazoid, 2024-07-13:
Open source of libbambu_networking.so for aarch64 and armhf would allow raspberry pi to run bambu studio.

HakunMatat4, 2024-09-05:
> However, it turned out that we were fooled.

You guys are lucky!!!

When I got my X1C a few months after release, the only way we could send files to the printer without being forced to use their cloud, was via FTP which https://github.com/bambulab uses eFTP IIRC violating its licence.

We not even had a Linux version of the slicer, we had to manually compile our slicer until the Linux community got together to get an AppImage working which for instance, the windowing has been upgraded and I cannot run the latest version of OrcaSlicer without upgrading my entire system.
Imagine that, you release a product Linux based and no Linux support, go figures.

BambuLab fooled many people, its printer ISN'T and will never be truly open-source.
That and tons of problems are the reason my X1C lasted 3 months I have since moved to Voron 3D printers.
I use OrcaSlicer and unfortunately, we are still forced to use this network plugin.

I do have firewall rules to block anything `bambu*` related so before any OrcaSlicer update, I must disable the firewall rule, update it, enable the rule again.

iCMDdev, 2024-09-05:
@HakunMatat4 
> https://github.com/bambulab uses eFTP IIRC violating its licence

I think the network plugin actually uses libcurl for FTP

Natureshadow, 2024-09-05:
>> https://github.com/bambulab uses eFTP IIRC violating its licence
>
>I think the network plugin actually uses libcurl for FTP

I stopped reading at the point where it uses FTP at all.

HakunMatat4, 2024-09-05:
> I think the Bambu Studio network plugin actually uses libcurl for FTP, but I might be wrong.

Nope haha
I remember logging to the printer and it does show you the welcome message like what happens with many FTP and SSH services.
I actually found the script I used to use.

```
#!/bin/bash

set +x

FILENAME=$1

expect <<EOD
spawn ftp 192.168.1.36
expect "Name (192.168.1.36:USER):"
send "bblp\n"
expect "Password:"
send "4ee01070\n"
expect "ftp>"
send "put $FILENAME\n"
expect "226 Transfer complete."
sleep 5
send "exit\n"
expect "221 Goodbye."
interact
EOD

```

And I would send the file like, to avoid being forced to use the cloud.
There was no other way to use the local network.
I believe X1C folks now do have a true LAN mode and can still use the webcam and all, we couldn't
I remember a dude in New Zealand who created a Python script and all haha

`send.sh Bottom.gcode.3mf`

-----

> I stopped reading at the point where it uses FTP at all.

It gets worse, unless things changed which I doubt, their cloud used to crash from time to time so folks who had spammed the slicer while the cloud was down, had their printers auto-printing the entire queue without any way to delete them without cancelling them manually.

-----

So anyway, the thing you are trying to solve now, we crossed this almost 2y ago when I had my X1C for 3 month.
Knowing BambuLab, I don't think you guys have much alternatives.
I crossed this ticket on OrcaSlicer Discord and got curious lol

iCMDdev, 2024-09-05:
> Nope haha I remember logging to the printer and it does show you the welcome message like what happens with many FTP and SSH services. I actually found the script I used to use.

Just to clarify, I was talking about Bambu Studio's network plugin (client FTP), not the printer (LAN / Cloud) FTP. Bambu says on their [open source website](https://wiki.bambulab.com/en/knowledge-sharing/open-source-software) that the X1 series uses `vsftpd`. 

> I remember a dude in New Zealand who created a Python script and all haha
>
> `send.sh Bottom.gcode.3mf`

The project is called X1Plus now AFAIK? From what I could find, this was an early version of it / the original exploit.

mikejhill, 2024-12-26:
I think this deserves more attention. From what I can tell, @ClayShoaf above is correct about @bambulab violating AGPL with this, which is extremely shady. I wonder whether this would be suitable for legal action from Prusa and/or Slic3r.

I am currently shopping around for printers. BBL units are at the top of my list, but OSS is critical for me, and this issue adds concern for the organization's trustworthiness. As it stands, using Bambu Studio over LAN would mean overlooking this ethical issue and other factors involved, including limiting our ability to conduct private security reviews.

ProfFan, 2025-01-19:
Under AGPLv3 anyone ever committed to the @prusa3d or @slic3r repos is entitled to sue under US copyright laws.

iCMDdev, 2025-01-19:
fyi: https://blog.bambulab.com/agpl-compliance-of-bambu-studio/

gooselord-0, 2025-01-19:
Brand new owner of a P1S, and I really wish I'd known about this before I bought the printer. It's extremely disappointing, and there's honestly no justifiable explanation for a closed-source network plugin, given the potential sensitivity of the operations. Bambu's response in the blog above doesn't address anything other than the its lawyers' opinion on the legality of the plugin's use under its license, which doesn't do anything whatsoever to mitigate security concerns.

Natureshadow, 2025-01-19:
Isn't that blog post an outright lie?

AFAIU, the network plugin is also necessary to comnunicate with the printer locally.

That said, any claims about the plugin violating the AGPL, of course, are nonsense. This topic is not a legal issue.

iCMDdev, 2025-01-20:
> AFAIU, the network plugin is also necessary to comnunicate with the printer locally. 

Yes, but afaiu their argument is that you can also simply slice your model and copy it on the SD card, thus an optional plugin.

iCMDdev, 2025-01-21:
Just in order to be clear, I was bringing up their point for reference, I do not endorse or support any side whatsoever, since I'm not a lawyer.

Out of genuine curiosity, what are your concerns over the plugin? Let's say, in a LAN setup with no internet access allowed to the printer (and perhaps the Bambu Studio plugin as well), what concerns you the most? (not an ideal setup, I understand)

Natureshadow, 2025-01-21:
>Out of genuine curiosity, what are your concerns over the plugin? Let's say, in a LAN setup with no internet access allowed to the printer (and perhaps the Bambu Studio plugin as well), what concerns you the most? (not an ideal setup, I understand)

I want to be able to analyse and debug all software I use.

But what concerns me the most is that BambuLabs wasn't honest to me. Their marketing claim is that their printers fully work without their cloud and with open source software. Both is not true.

makkarpov, 2025-01-21:
>Let's say, in a LAN setup with no internet access allowed to the printer (and perhaps the Bambu Studio plugin as well - perhaps this is the hardest one to control from your perspective), what concerns you the most?

The fact that this setup does not work unless printer and computer are in the same broadcast domain. And BambuLab has apparently no interest in ever fixing that. You cannot type in the IP directly, it has to be auto-discovered by their magic binary blob.

iCMDdev, 2025-01-21:
> I want to be able to analyse and debug all software I use.

Definitely not a replacement for open source software, but, if you really find yourself liking to research stuff like this, you may consider using a decompiler. Everything is "open source" with it :) 

But this kind of research will always take some time...

Natureshadow, 2025-02-24:
> (and [disadvantages](https://en.wikipedia.org/wiki/XZ_Utils_backdoor)).

Just so noone falls into this trap:

The only consequence XZ being open source software had in the referenced case is that the backdoor could be found and proven. Andreas, the person who uncovered it, could not have done so if XZ had been a proprietary product. The infiltration by the attackers, however, would have been very much possible if it had not been open source software, and most likely it would have been much easier.

Natureshadow, 2025-02-24:
>I don't know if it would've been easier to develop exploits if the software was closed - I'd say it's even harder because it's harder to understand what your project does. But this also makes fair, ethical research, harder too, as you already said. 

I think it is much easier to get hired into a senior position by a proprietary vendor than it is to infiltrate something publicly while the whole world is watching.

iCMDdev, 2025-02-24:
> I think it is much easier to get hired into a senior position by a proprietary vendor than it is to infiltrate something publicly while the whole world is watching.

Yup, you’re bringing a good point here as well!

That is why I think independent security researchers are really important in general.

dyfrgi, 2025-03-28:
I want it to be open sourced so that I can actually run it - the plugin they ship is not compatible with NixOS. I'm frustrated to discover that I cannot use my printer without either running a different OS or running back and forth with SD cards. I was previously using it via Windows, but I no longer have a Windows system.

obsm-maker, 2026-01-07:
@lanewei120 This is unacceptable. Users should be able to use any os they choose. Either your software is open source or not.

ClayShoaf, 2026-10-11:
Maybe someone should go ahead and use a frontier model to open source this for them...
https://github.com/morluto/rea

Natureshadow, 2026-10-11:
>Maybe someone should go ahead and use a frontier model to open source this for them...

No thanks. We don't fuck corporate using corporate fuck.

cgarz, 2026-11-04:
If AI is good enough to forcibly open source the Adobe suite then it's good enough for ScambuLab.

It's only fair we use the tools of scumpanies against them.
Hats off to anyone who takes this on. Keep us updated.