bambulab/BambuStudio #2383: 几个简单的小建议

Dong-Tu, 2023-08-25:
1.预览支撑所消耗耗材的重量。
目前studio中似乎无法直接显示支撑消耗耗材的重量，只能通过切片有支撑和无支撑的模型来推算支撑的消耗。
2.清理喷头时，先擦嘴再冲刷。
目前的流程是冲刷耗材后才会擦嘴，这会导致一个问题：喷嘴内残余的耗材会从喷口流出并拉成一条丝，这条丝的长度过长时会挂在废料漏斗外侧，阻碍废料的正常排出。在使用ams进行多次换料时这个问题更是可能直接导致耗材堆满漏斗进而导致打印失败。
3.让用户可以选择打印成功后自动关闭LED灯
4.自动检测监控画面卡住的状态并刷新，或者在studio的“设备”页面新增一个“刷新监控画面”按钮。

SaltWei, 2023-08-25:
> 目前studio中似乎无法直接显示支撑消耗耗材的重量，只能通过切片有支撑和无支撑的模型来推算支撑的消耗。

建议很好，接受

SaltWei, 2023-08-25:
> 清理喷头时，先擦嘴再冲刷
这个需要设计换料过程的同事确认，我们内部会转达

SaltWei, 2023-08-25:
> 监控画面卡住时自动刷新，或者在studio的“设备”页面新增一个“刷新监控画面”按钮

@bambu123 请评估一下

SaltWei, 2023-08-25:
> 让用户可以选择打印成功后自动关闭LED灯

你是想打印机上有个选项，勾选之后，每次打印结束都自动确保关闭LED灯吗

bambu123, 2023-08-25:
@Dong-Tu 你是不是有 P1P 机器 Liveview 播放一会儿画面静态问题，是不是 Studio 1.6 版本没有这个问题，从 1.7 版本开始有这个问题？

pichen2021, 2023-08-25:
> > 清理喷头时，先擦嘴再冲刷
> 
> 这个需要设计换料过程的同事确认，我们内部会转达 @pichen2021

因冲刷前喷嘴可能已经溢出一截料，擦嘴可能挂在喷嘴上导致异常；若需修改动作，可在studio中按如下步骤在冲刷前增加擦嘴动作：
[image]

Dong-Tu, 2023-08-25:
> > 让用户可以选择打印成功后自动关闭LED灯
> 
> 你是想打印机上有个选项，勾选之后，每次打印结束都自动确保关闭LED灯吗

是的，不过这个属于可有可无，没有也不要紧

Dong-Tu, 2023-08-25:
> @Dong-Tu 你是不是有 P1P 机器 Liveview 播放一会儿画面静态问题，是不是 Studio 1.6 版本没有这个问题，从 1.7 版本开始有这个问题？

是的，1.6版本几乎没有出现过这个问题，现在不仅是P1P，另一台P1S也会出现

SaltWei, 2023-08-25:
> > > 让用户可以选择打印成功后自动关闭LED灯
> > 
> > 
> > 你是想打印机上有个选项，勾选之后，每次打印结束都自动确保关闭LED灯吗
> 
> 是的，不过这个属于可有可无，没有也不要紧

@sodachen22 请评估该功能并加入用户需求池

Dong-Tu, 2023-08-25:
> > > 清理喷头时，先擦嘴再冲刷
> > 
> > 
> > 这个需要设计换料过程的同事确认，我们内部会转达 @pichen2021
> 
> 因冲刷前喷嘴可能已经溢出一截料，擦嘴可能挂在喷嘴上导致异常；若需修改动作，可在studio中按如下步骤在冲刷前增加擦嘴动作，保存后重新切片： [image]

非常感谢解答

leoleoasd, 2024-01-01:
> 因冲刷前喷嘴可能已经溢出一截料，擦嘴可能挂在喷嘴上导致异常；若需修改动作，可在studio中按如下步骤在冲刷前增加擦嘴动作，保存后重新切片： [...]

我的A1也有这个问题，是否有必要统一修复？
冲刷前喷嘴上已经有了一截料，冲刷时那节料会跑到那个冲刷的地方的 竖着活动的板子（不知道叫啥）的右边，导致冲刷出来的料掉在热床上