bambulab/BambuStudio #2385: 3Dconnexion Wireless Space Mouse mouse capture issue

ukdavewood, 2023-08-25:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Ventura 13.4.1 (c) 

### Additional system information

Mac mini M1 16gb

### Printer

X1C

### How to reproduce

Start Bambu Studio with a blank project
Add a primitive - such as a cube
Spin the cube with the Space Mouse - it doesn't spin that well - but just about
After doing this the left click no longer works on the normal mouse in any application - Main mouse is a Logitec MX Master Mac - but also when I try another mouse it also doesn't work too.

### Actual results

Mouse seems to be captured by Bambu Studio and left click no longer works - rendering Mac useless - and forcing a hard shutdown - losing most changes in BS and other applications.

This has happened to me numerous times recently - but only on most recent versions of BS - previous versions didn't cause this issue.

I see an 3dconnextion issue was recently closed - so I assume some changes were made.

### Expected results

Mac mouse doesn't get taken over - and space mouse and Mac works ok - like it used to, and like it doesn't in prusa slicer for example.

As BS doesn't crash I haven't included a crash log etc and you don't need a project file.

I have still got a few older versions of BS installed - so think I will try a few different ones until I find one that doesn't cause the issue.

### Project file & Debug log uploads

[Archive.zip](https://github.com/bambulab/BambuStudio/files/12440984/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ukdavewood, 2023-08-25:
Updated - have tried all of the following versions of BS - and the issue occurs on all of them - I just need to load a project file with a cube in it, try to spin it, and then after that Mac mouse doesn't work.
8-1.7.4.52
7-1.7.3.50
6-1.7.2.51
5-1.7.1.62
4-1.7.0.65
3-1.7.0.53
2-1.6.1.55
-1.6.2.4

ukdavewood, 2023-08-25:
Further update - just changed my 3dconnextion settings for Bambu Studio to disable for everything - tested on 1.7.4.52 - Mac doesn't lose mouse ability now - so useful to know.

Obviously the 3d mouse doesn't work with Bambu Studio with this setting though - but at least it doesn't crash to Mac.

bambu123, 2023-08-29:
@ukdavewood  Does prusa have such a problem? 
According to what you said about the capture mouse, which UI sub-window do you think caught the mouse, such as the 3D window?

ukdavewood, 2023-08-29:
No Prusa slicer still works ok.
The mouse cursor doesn't get stuck in the 3d window - it's just that the left click stops working - the keyboard does still sort of work - so you can for example force quit BS, but you still don't get your mouse button back. 

Will try it on my Mac laptop to see if I get the same problem with mouse and trackpad