bambulab/BambuStudio #2386: Add the ability to monitor and display amount of filament remaining on a spool

julie777, 2023-08-25:
## Is your feature request related to a problem? Please describe.

Sometimes I do not have enough filament remaining to print a part, and I do not have another spool of that filament available to be able to swap spools mid print. I currently have to make a guess whether to try the print and hope for the best, or just wait until I buy more filament.

Another use case is when printing overnight or remotely and I do not want to start a print that cannot finish.

Note: I do not have an AMS, and even if I did the AMS does not track non-BBL filament without RFID tags.

## Describe the solution you'd like

In Bambu Studio provide the ability to number filament spools and have it use the estimated amount of filament for a model to track usage for that spool.

The implementation of this is simple and does not require being extremely accurate to be useful.

### Supporting Structure

Add a numbered spool list to the Bambu Studio configuration to keep track of the known spools and current weight.

### User Interface
1. In the print dialog provide spool number drop-down menu to allow choosing which spool is currently being used. The default value should be the last spool used. (It is common to print multiple jobs one after another using the same spool.)
2. The last option in the menu should be add new spool which provides a way to enter the starting weight.
3. In the chooser, the current calculated weight should be shown next to the spool number. 
4. if the weight for the current spool is less than the amount needed to print then the user should be warned before printing. However, the user should be able to continue if desired. The printer will run out, stop and notify the user. The user will load a new spool. When printing continues, Bambu Studio should prompt for the spool number of the inserted spool and continue based on that spool.
5. allow the user to remove a spool number from the list, or reset the weight (if reusing the same number is allowed)

### Notes
- because it might be difficult to know how much filament has been use up to the point where the filament runs out and is changed by the user, that part could be optional. However, given that the printer is reporting the current layer number, I believe that Bambu Studio can calculate this.
- better accuracy can be achieved if the estimated amount of filament for a print job includes any required filament during the startup code. I do not know if that is in the estimated amount of filament currently.
- Optionally, provide the ability to set the current spool number for a given printer in the "prepare" section so I can just set it after I change spools and not have to wait until I am sending the next print.

## Describe alternatives you've considered

I could weight my spools full and empty and keep my own list and update it manually.

SnickCH, 2023-08-26:
+1
It shouldn't be to hard to calculate how much it is left. Maybe it would be easier with the length because then it is not important about the density of the filament
- initial weight or length (set by the user)
- bambustudio estimates the weight / length of the print (+ some margin) and removes it from the current weight / lenght

ShawnLeary, 2023-10-02:
Would love if AMS had a general tracker setting.  If new spool it should have an idea of how much is left.

EA0003, 2024-03-23:
Yes! this would be great, specially since the weight is really what matters. Even if they add a "safety" factor of 5% which the user can adjust, that'd be so nice

theukedge, 2024-04-08:
+1. Especially with AMS, Bambu Studio ought to be able to track how much filament is left on each spool and allow the user to view that information and/or provide a warning if you're trying to print something that you don't have enough filament for.

cfuller007, 2024-04-16:
I would love this. I would love to track how much each print cost me in price and filament. Additionally, this would allow me to be ready when I need to order more filament from bambú and how much I have left.

PnsDev, 2024-06-09:
huge +1

MathiasSven, 2024-07-31:
+1! But also, isn't this somewhat a duplicate of #1186

mike-masson, 2024-08-02:
> +1! But also, isn't this somewhat a duplicate of #1186

+1 

Yes, but that seems like an idea in which they create an entire indexing database, which would be awesome and the CEO did mention something like this in the AMA that he did over on reddit. Whereas this seems more like a quick addition as it should be a relatively easy thing to do since Bambu Studio already estimates remaining filament by measuring how much filament is pulled per rotation of the spool. I'm not sure what the margin of error is, but even 10% would be useful.

jjvelar, 2024-11-04:
Hi,
This will be a great adding for AMS Lite.
Is this in the pipeline?

wawibu, 2024-12-06:
+1 
I'm new to Bambu and would like to see the remaining length. Others - like Octoprint - could provide that information. No need to be 100% accurate, I would love to get an indication on how much is left so that I know before I print if that spool would have enough or not

FaberO42, 2025-01-01:
Big +1 for something along these lines.
Bambu great for managing 3D printing process, but significant room for improvement in managing key (filament) consumables use.
Also requires to support more than just Bambu brand filaments.

chunkahunk, 2025-06-25:
+1 please