bambulab/BambuStudio #2387: Calibration/calibrate button on title bar disappears

Thibo-do-it, 2023-08-25:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11 64 bit

### Printer

X1C

### How to reproduce

1. open bambu studio
2. go to prepare tab
3. sometimes the calibrate/calibration button is in the title bar of the window, and other times it is missing

### Actual results

sometimes the calibrate/calibration button is in the title bar of the window, and other times it is missing

[image]

### Expected results

calibration button should always be there

### Project file & Debug log uploads

not sure when it last appeared

### Checklist of files to include

- [ ] Log file
- [ ] Project file

MadMax389, 2023-08-26:
Calibration only shows up on the top toolbar when you have a non-Bambu printer selected. 
[image]

Thibo-do-it, 2023-08-26:
It also shows up sometimes when my X1c is selected. I don't have another
printer setup on bambu studio.

How can I access all the calibrations when the X1c is selected?

Regards,
Caleb Thibodeaux

MadMax389, 2023-08-26:
It should show up on the lower toolbar next to Project
[image]

Thibo-do-it, 2023-08-26:
That calibration tab is missing most of the calibration tests such as max
volumetric flow rate and temp tower.

Regards,
Caleb Thibodeaux

MadMax389, 2023-08-27:
Those particular calibrations are only available in Orca, I believe.

Thibo-do-it, 2023-08-27:
That's the issue. They are not only in orca. I created the bug because
they're available in bambu studio sometimes.

Regards,
Caleb Thibodeaux

MadMax389, 2023-08-27:
The only time I have ever seen that calibration tab on the top toolbar is when there is a non-Bambu printer selected. (There is an issue thread on this somewhere in Github. Like https://github.com/bambulab/BambuStudio/issues/2101) Even when it is available, the gcode does not work correctly for the temp tower. The temperatures are too low. My point is it's not a bug, it's just not a supported feature yet.

[image]

Thibo-do-it, 2023-08-27:
I see. Thank you for the clarification. That makes sense. My calibration
towers were coming out weird.

Regards,
Caleb Thibodeaux

MadMax389, 2023-08-27:
I just used Orca to do the temp tower and maximum flow calibrations, which are one-time tests per filament material as far as I'm concerned. The automatic flow ratio and flow dynamic calibrations (k-factor) available in Studio for the X1C are easier and better than the Orca tests in my opinion. I use the values generated on the X1C in my filament profiles for my P1P.

Regards

zhimin-zeng-bambulab, 2023-08-30:
When you select a non-Bambu preset, the calibration button in the title bar will show, and it will not show in toolbar next to Project. And if you select a Bambu preset, it will show in toolbar next to Project.

At present, there are only "Flow Dynamic" and "Flow Rate" two calibrations for bambu. If there is any need for other calibrations, you can discuss them there(https://github.com/bambulab/BambuStudio/issues/2101).

closed 2023-09-01 by zhimin-zeng-bambulab

mmmlinux, 2023-12-20:
I can repeatedly make this configuration menu appear with a bambu x1c selected if i drag bambu studio from a high dpi display to a lower dpi display, It will cause the config menu to appear. Please make it available with our bambu brand machines!

manofpants, 2024-06-23:
This is still happening. Newest version of Bambu Studio 1.9.3.50 and I can't see the calibration menu half the time.

MULAWI, 2024-12-24:
> I can repeatedly make this configuration menu appear with a bambu x1c selected if I drag bambu studio from a high dpi display to a lower dpi display, or the other way. Then the config menu will appear. Please make it available with our bambu brand machines!

Thats true, if I move the Bambu Studio to a less reslution Display it desapears and back again if I move it to a higher resolution Display.

Sveeeeeen, 2025-04-20:
Why is this still an issue in 2025 and v2.0.2.57?
Please fix it.

ryanmiller831, 2025-05-20:
I am still getting this bug as well. Can Bambu developers please include the manual calibration feature for advanced settings in Bambu Studio? If it is available for other printers, why can't we use it as well?

ITBeee, 2025-06-28:
Same Issue. I use the Bambu Studio V 2.1.1.52. The Calibration Tab only shows up if i turn my Monitor off and then on again.