bambulab/BambuStudio #2388: layer progress not updating on device tab

geoffpedder, 2023-08-25:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1P

### How to reproduce

start a print, go to device tab and look at printing progress. 
for example, i'm printing a thing with 17 layers and it never updates which layer it's on, it always sats  - Layer 0/17 
It happens the same on all prints

### Actual results

[image]

### Expected results

.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-08-30:
Hi. @geoffpedder 
Can you provide the model file you printed

walterwongbbl, 2023-09-01:
> @walterwongbbl I was about to create a bug report too but noticed this issue plus #2372 and #2321 all appear to be related.
> 
> It's been a constant issue for me personally. This usually happens for one of two reasons in my experience:
> 
> 1. (the most common) Macbook Pro goes to sleep, upon waking it won't reconnect to the printer.
> 2. Network changes, drops or occasionally during VPN connects/disconnects (no LAN change)
> 
> Both have appear to cause same result. The progress freezes and call buttons to stop/pause etc are clickable and the UI popup appears but do nothing to the printer.
> 
> Currently having this in a calibration where it printed everything and I know the K factor but now I'll have to rerun the whole calibration and reprint it just to get to the screen to type in the K Factor. Would be great if there was also the ability under "Manage results" to add in new profiles you know the K Factor for.
> 
> This would help solve this issue for me personally and allow people to migrate Pressure advance settings from Orca back to Bambu which is a bit of a reason I'm sure people end up having to stick on Orca even after you add feature parity.
> 
> It's less of an issue on regular prints IMHO - more just an annoyance - as they'll complete fine still and don't require UI followup like the calibration tests do.
> 
> @geoffpedder @DarrenLynch @dguinaldo would be great to have your input too if the circumstances are similar.
> 
> ps. @geoffpedder are you the same Geoff at ILM who did the May 4th talk at the MacMillan Space Centre?

I will convey your needs to my colleagues

walterwongbbl, 2023-09-01:
@joshmeads 
Please create a new feature for calibration and we will handle it separately.

geoffpedder, 2023-09-01:
hi, I was printing the pixel plate walls for the p1p from bambu. I have since noticed the layer number is working on other models though. I did actually work at ILM up until May, but I didn't do a talk at MacMillan Space Centre :)

walterwongbbl, 2023-09-05:
> hi, I was printing the pixel plate walls for the p1p from bambu. I have since noticed the layer number is working on other models though. I did actually work at ILM up until May, but I didn't do a talk at MacMillan Space Centre :)

 It might be an issue with the model file. You can choose to upload the problematic model file, and we will investigate the cause, or you can choose to close this issue.

walterwongbbl, 2023-09-19:
Hi.  @geoffpedder 
Have you encountered the same problem recently? If not, this issue will be closed.

geoffpedder, 2023-10-25:
no all good. sorry for the late reply

closed 2023-10-25 by geoffpedder