bambulab/BambuStudio #2389: Preview Tab:  Synch Top Visible Layer with Live Layer Value

Chireven, 2023-08-26:
**Describe the solution you'd like**
There are some prints that I want to babysit, sometimes at a particular layer.  To accomdate these, I've found that I switch between the Preview tab and the Deisgn tab.  I retrive the current layer from the Design tab, then review the layer in the preview tab using the top visible layer slider.   This feature would, while active, keep the top visible layer height in synch with the layer height on the design page.

The design page already has the live layer, so it should be accessible via code already.
[image]
  
The UI already supports changing the value of visible top layers.
[image]

A checkbox that linked these values is the only thing missing.

**Describe alternatives you've considered**
The only alternative that I have considered is doing this manually. 

**Additional context**
I know this is a "quality of life" feature, but I believe it would be a simple addition that wouldn't really take away from any other enhancment.  I wouldn't want this to be prioritized over a serious feature, but I believe this would be an interesting way to monitor those jobs that just need a little extra attention.

Thank you for considering this request!

1RV34, 2026-01-02:
I would love this too, it's something I've used in other slicers but am missing from Bambu Studio.

Edit: For reference, I've seen other slicers call it "Live Preview".

1RV34, 2026-01-02:
One issue that I do possibly see arise is that the sliced model isn't locked, so people could re-slice an adjusted version or a totally new model and try to live-sync with the wrong sliced model, or a new empty process is opened and people wanting to sync.

I have thought about some ways to deal with this though.

First of, when sending a print the Live Preview would be nicest to be immediately enabled. Then, when trying to change any configuration or model while the option is enabled, a confirmation popup could show. That way we prevent accidental de-syncs.

Secondly, when having re-opened a 3mf and trying to enable the sync, there could be some kind of checksum check, if same 3mf as is printing go ahead and enable, if not, show modal saying it's the 3mf file doesn't match. (This solution assumes there is a way to get a checksum from the printer, I don't know if there is?)

Lastly, if trying to enable while the wrong 3mf is open or an empty 3mf is open, could the slicer allow the user to download the 3mf file from the printer? (again, if there's a way to download the model from the printer?)

I'm currently looking into the printer's capabilities myself and I find some search results that there's a way to download a `.gcode.3mf` sliced model file from the printer, which would make things a lot more convenient.