bambulab/BambuStudio #2393: Error in top layers on 1.7.4.52

DamoclesSword, 2023-08-26:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Create a print with any profile as long as setting **Only one wall on top surfaces** to "All top surfaces" with Monotonic Line as the top surface pattern, and Arachne are enabled. 
2. Inspect the top layer. 

I have tested this with several calibrated filaments (flow and PA) and the issue is universal. If these are turned switch off/to a different pattern, the issue goes away. 

### Actual results

When setting **Only one wall on top surfaces** to "All top surfaces" with Monotonic Line as the top surface pattern, the top surface is very rough and uneven. 

### Expected results

The surface should be smooth. If you disable the one wall and switch to Monotonic for the top surface, the issue goes away completely. 

### Project file & Debug log uploads

N/A 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2023-08-28:
we will check that.

BambooBugFixer, 2023-09-07:
Hi, I did a test with 8 cubes. The setting is like below:
[image]
The top surface of each cube is smooth and fine.
[image]
Maybe you can share your model. We'll do more testing.

SaltWei, 2024-01-16:
Does this issue still exist? Could you share your file for us to reproduce?