bambulab/BambuStudio #2395: The Textured PEI Plate is omitted from the Flow Dynamics

edquichan, 2023-08-27:
### Bambu Studio Version

01.07.04.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1 Carbon with AMS

### How to reproduce

1. Select Flow Dynamics
2. Select Plate Type
3. Textured PEI plate is missing from the dropdown.

### Actual results

When you select Flow Dynamics, the Textured PEI Plate is missing from the dropdown. 

### Expected results

The Textured PEI Plate should be included in the Flow Dynamics dropdown.

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

MadMax389, 2023-08-27:
The lidar on the X1C doesn't play well with the textured plate and won't give repeatable results. I use the cool plate or other 3rd party smooth plate to generate the k-factor profiles. Once generated, they should work when using the textured plate.

zhimin-zeng-bambulab, 2023-08-28:
@edquichan 
The automatic mode needs to use the lidar which cannot work well with the Textured PEI plate.
So in this case, you can use the manual mode instead.

closed 2023-08-31 by zhimin-zeng-bambulab