bambulab/BambuStudio #2396: "Text shape" broken in 1.7.4.52?

j0nas, 2023-08-27:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10/macOS

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. In "Prepare view", click on "EmptyLabel" printable object
2. Click on "Text shape"
3. Enter text in "Input text" field
4. Click somewhere on printable object

### Actual results

[image]

A small magenta-colored square appears. Attempting to click on it simply focuses the object.

[image]
After playing around with this a lot more, I found a couple pixels where the text appears. It seems to only want to place the text on the side of the object.

### Expected results

The preview text should appear and be placable no matter which surface I hover.

### Project file & Debug log uploads

[Text bug.zip](https://github.com/bambulab/BambuStudio/files/12447557/Text.bug.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

j0nas, 2023-08-27:
More context: if I rotate the "EmptyLabel" object so that it is upright, I can place the text. But I cannot do so when the object is lying down. This will likely be very confusing for people who don't know how to work around this. 
[image]

My best guess is that the "text shape" tool works on the objects **initial** orientation instead of its current orientation; the "EmptyLabel" object was imported upright. 

Placing the text in the initial position and then rotating the object works as a workaround.
[image]

j0nas, 2023-08-27:
Editing the text does not work either, but works again if I rotate the object back to its initial orientation.

MadMax389, 2023-08-27:
I have also noticed some weirdness around text orientation vs initial model orientation. If you uncheck the "Surface" box, you can lay it flat on the top surface, however.

zhimin-zeng-bambulab, 2023-10-08:
It has been fixed in V1.7.6

closed 2023-10-11 by zhimin-zeng-bambulab