bambulab/BambuStudio #2402: Sleep mode MAC, update printing process

DJ-O, 2023-08-27:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS 11.7.8

### Additional system information

[image]

### Printer

BL X1C

### How to reproduce

When the printing is finished and my MAC is in "sleeping mode", I can't start a new print job as Bambu Studio is still assuming the printer is still busy. I have to close and then open Bambu lab for the refresh. Very annoying. Older versions it was never a problem.

### Actual results

[image]
[image]

### Expected results

The printer needs to send feedback to bambulab studio that the printing job is finished. Or please make a refresh button in the tab "device" or something. 

Thanks 
Brgds Didier

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-08-30:
@DJ-O 
Hi.
We are trying to solve this issue.

walterwongbbl, 2023-09-07:
@DJ-O 
Does this issue occur every time your computer wakes up from sleep mode?

walterwongbbl, 2023-09-07:
@DJ-O 
Does Bambu Studio freeze or crash when your Mac wakes up from sleep mode?

DJ-O, 2023-09-07:
Hello Walter.

No, Bambu studio never freeze or crash. I can do anything except starting a
print job (via cloud). During printing, BL studio gets feedback from the
printer as you can see at the progress bar. After printing apparently not
anymore and BL studio “things” the printer is still busy. The progress bar
doesn’t go to 100% but stays at a value when the Mac go’s to sleep I guess.
The strange thing is the camera image works live, so printer send data.

Brgds Didier

Met vriendelijke groet, Didier

walterwongbbl, 2023-09-07:
Hi. @DJ-O 
Can you provide the log files from when the issue occurs?
[image]
[image]

walterwongbbl, 2023-12-21:
We have applied the latest version, please download the latest version to experience it.

closed 2023-12-21 by walterwongbbl