bambulab/BambuStudio #2404: MQTT connection to printer is no longer reliable - connects but receives no data

AdrianGarside, 2023-08-27:
### Bambu Studio Version

1.6.2.4

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P

### How to reproduce

1. Connect to printer by MQTT Explorer / Home Assistant integration.
2. Some of the time it will connect sucessfully but never receive any data back
3. In that scenario, you can reconnect as many times as you like - it won't get data until the printer is restarted.
4. And then it's hit and miss if it will get data. If it gets data it seems like it will continue to get data indefinitely. If it doesn't get data only a printer restart may fix that.

Seems likely to be related to:
"Fixed the printer restart problem caused by continuous reconnection of MQTT using third-party tools." - see https://wiki.bambulab.com/en/p1/manual/p1p-firmware-release-history
As the issue started with the P1 series OTA version01.04.00.00（20230807）firmware update.

### Actual results

If mqtt data doesn't flow only a printer restart can fix it but not consistently.

### Expected results

Successful MQTT connection should always result in receiving MQTT data on the report topic.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [X] Log file
- [X] Project file

AdrianGarside, 2023-08-27:
This affects Bamby Studio too it seems - it's no longer detecting my printer where Orcas Slicer, one of my HA instances and Bamby Handy are able to see it. Updated to 1.7.4.52 and it's still not listing my printer.

AdrianGarside, 2023-08-29:
From more investigation the behavioral problem is simple - only the most recent connection gets the data. But it's a stack. So if I connect my home assistant instance it will work until I also start up mqtt explorer at which point only the latter gets data but once I disconnect that my home assistant instance will start getting data again. This breaks my ability to use a separate test instance of home assistant to work on the integration since it breaks my production instance.

slynn1324, 2023-09-01:
I've noticed the same problem on my P1S with the latest firmware.  It's not the most elegant solution, but as a workaround I hacked in a webhook into my fork of BambuStudio to echo out the received status information so I can ingest it into Home Assistant --> https://github.com/slynn1324/BambuStudio/commit/d919b7bdc5d72e870ba60f77d85b5de6e9fcd232.   This puts the extra load on the PC running BambuStudio rather on the printer.  I don't really propose adding this as a main feature... but wasn't sure how else to get it working for my case.

AdrianGarside, 2023-09-03:
This also breaks lan mode usage by multiple clients (e.g. Bambu Studio and Orca Slicer).

npike, 2023-09-06:
Is there a version of the P1 firmware that we can roll back to that works as intended?

sodachen22, 2023-09-09:
Does the latest Bambu Studio support P1S LAN only mode?

AdrianGarside, 2023-09-09:
> Is there a version of the P1 firmware that we can roll back to that works as intended?

01.04.00.00 introduced the bug. Earlier firmware versions behaved ok.

gustafssone, 2023-09-22:
> I've noticed the same problem on my P1S with the latest firmware. It's not the most elegant solution, but as a workaround I hacked in a webhook into my fork of BambuStudio to echo out the received status information so I can ingest it into Home Assistant --> [slynn1324@d919b7b](https://github.com/slynn1324/BambuStudio/commit/d919b7bdc5d72e870ba60f77d85b5de6e9fcd232). This puts the extra load on the PC running BambuStudio rather on the printer. I don't really propose adding this as a main feature... but wasn't sure how else to get it working for my case.

Interesting workaround. That in combination with a slicer that is always running, e.g. https://hub.docker.com/r/mikeah/orcaslicer-novnc could work for now. How do you access the webhook? Got any more details?

TackleBerryCrash, 2023-09-27:
Will this bug be addressed in a short time frame ? 
I was hoping it would have been with new fw update but it is still replicable

AdrianGarside, 2023-09-27:
It's out of our control since it's a bug in the P1P firmware. The only workaround is to not use lan mode and configure this integration to connect to bambu cloud mqtt.

AdrianGarside, 2023-11-02:
@sodachen22 is there any hope this will be fixed soon? It breaks lan mode configuration and affects integrations like the home assistant ones and XTouch (formerly P1Touch).

sodachen22, 2023-11-02:
Thank you for your feedback. We have identified a software issue here. The issue occurs when multiple MQTT clients are connected, causing only the last connection to receive data. This problem will be fixed in the upcoming versions. The temporary solution for now is to keep only one connection."

AdrianGarside, 2023-11-21:
@sodachen22 do you know if it is slated to be fixed in the next firmware update? This continues to generate a lot of complaints.

xperiments, 2023-11-24:
Yes , we have a lot of user reports with this also in xtouch
Any ETA?
We need to provide a "bridge" solution?
Thanks

kenkamm, 2023-11-26:
@sodachen22  Any updates on this? Would really like to see this fixed in the next update if at all possible.

raster, 2023-11-29:
MQTT needs to work reliably. Hoping this gets fixed soon.

smford, 2023-11-29:
This is a very frustrating bug, hopefully it gets fixed soon.

TackleBerryCrash, 2023-11-29:
This was working 2 versions ago so needs to be fixed ASAP. Telling users "Don't use multiple connections" is not a good solution :)

Sprawlie, 2023-12-14:
Is there an update on this and the latest version? On 1.04.02 and this is still occurring.

AdrianGarside, 2023-12-15:
Please say this is going to be fixed in the v1.5 firmware that's soon to be released.

Genygo, 2023-12-15:
The BambuLab support confirmed to me, that this will be fixed in the next release.
But the printer will only support to bind to 4 devices.

TackleBerryCrash, 2023-12-28:
> The BambuLab support confirmed to me, that this will be fixed in the next release. But the printer will only support to bind to 4 devices.

I guess it was the "next" next release...

AdrianGarside, 2023-12-29:
> > The BambuLab support confirmed to me, that this will be fixed in the next release. But the printer will only support to bind to 4 devices.
> 
> I guess it was the "next" next release...

Seems to be fixed in 1.0.5.0