bambulab/BambuStudio #2407: Infill adds Concentric areas

TheAtomicSoul, 2023-08-27:
### Bambu Studio Version

01.07.04.52

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

P1P

### How to reproduce

Set infill to "Monotonic" or "Renticular".
Some areas get concentric infill while others get the expected direction. 

### Actual results

any infill on parts that vary in size?

### Expected results

Each infill layer to follow the designated pattern. 

### Project file & Debug log uploads

na

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BambooBugFixer, 2023-09-07:
Hi, I failed to reproduce this issue. "Monotonic" and "Renticular" infill patterns work fine in my testing. Can you attach your model and screenshot the issue? It would help a lot.

TheAtomicSoul, 2023-09-07:
Sorry, I thought I closed this ticket! I found out turning off "detect narrow internal solid infill" solves this problem. 
I thought that was an option that applied to much smaller areas than it is. (Perhaps it needs a variable setting so it CAN only be applied to smaller areas.)

BambooBugFixer, 2023-09-07:
All right. When enabling "detect narrow internal solid infill" option, concentric pattern will be used to speed up printing.

closed 2024-01-16 by SaltWei