bambulab/BambuStudio #2408: Micro SD card read error

jay3x7, 2023-08-28:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 7 5800X 8-Core Processor                3.80 GHz
RAM: 16.0 GB
GPU: NVIDIA GeForce RTX 3070 Ti

### Printer

X1C

### How to reproduce

Go to micro SD on machine display
Select .3mf file with 12 files
Select the 10th file
Plate one is displayed and locks machine. 
Bambu studio shows machine as busy even after restart.
Fix to get out was to run self test from machine display.

### Actual results

Printer was able to be available to print again after doing the self test. Before, studio would show printer as busy even after restart. Printer itself showed as idle after restart. Selecting the 10th plate again would result in plate 1 again showing up and printer just sat. Would not start sequence to print anything.

### Expected results

Printer should see plate 10 as a separate print job and print plate. 

### Project file & Debug log uploads

[New WinRAR ZIP archive.zip](https://github.com/bambulab/BambuStudio/files/12449672/New.WinRAR.ZIP.archive.zip)
Project file is 56MB and unable to be dropped

### Checklist of files to include

- [X] Log file
- [ ] Project file

jay3x7, 2023-08-28:
Forgot to add firmware versions:
X1C 01.06.03.00
AMS 00.00.06.32

sodachen22, 2023-08-30:
@jay3x7 Could you please share the gcode.3mf with us that we can reproduce this issue? Or create a ticket with the LOG of the printer (https://wiki.bambulab.com/en/create-a-support-ticket)?

jay3x7, 2023-08-30:
Found that SD Card was corrupted and fully trashed. This issue is solved. Thank you for your time.

closed 2023-08-30 by jay3x7