bambulab/BambuStudio #2409: Flow Rate Calibration: New Filament Profiles Not Reflected

devnulled, 2023-08-28:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura 13.5

### Additional system information

Macbook Pro, M1Max, 64 GB of RAM

### Printer

X1 Carbon

### How to reproduce

1. Click On The "Calibration" Tab
2. Click On The "Flow Rate" vertical tab on the left
3. Under "Filament Calibration", select one of the AMS Slots, and then a filament profile
4. Click Back To the "Prepare" Tab, create a new Filament profile
5. Repeat process above; the new Filament profile you created does not show up as one you can do a flow rate calibration for

### Actual results

The new Filament profile you create does not show-up as one you can use.  The only way to be able to use it that I've found is to close the app and start over again.

### Expected results

The new filament profile should show up so that you can use it to do a flow calibration.

### Project file & Debug log uploads

Not using a project file.  Just trying to run a calibration.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-08-29:
@devnulled 
When you switch back to the Calibration page, you can re-select the nozzle diameter, which will refresh the filament list.

zhimin-zeng-bambulab, 2023-08-31:
@devnulled  Do the above steps solve your problem?

zhimin-zeng-bambulab, 2023-09-25:
> @devnulled When you switch back to the Calibration page, you can re-select the nozzle diameter, which will refresh the filament list.

If you encounter this problem, you can follow this step.

zhimin-zeng-bambulab, 2023-09-25:
1

closed 2023-09-25 by zhimin-zeng-bambulab

oilgas, 2024-02-22:
Work like a charm select again the nozzle !! Good discovery