bambulab/BambuStudio #2411: Multicolor issue with round corners

superfloh247, 2023-08-28:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 13.5

### Additional system information

Apple M1pro

### Printer

P1P

### How to reproduce

Apply multi color to object using the color tool in height range mode

### Actual results

See screenshots and .3mf file

[BambuBugReport.3mf.zip](https://github.com/bambulab/BambuStudio/files/12454632/BambuBugReport.3mf.zip)
[image]
[image]
[image]
[image]

### Expected results

all lines (outer and inner and infill) on layer 5 are black

### Project file & Debug log uploads

[BambuBugReport.3mf.zip](https://github.com/bambulab/BambuStudio/files/12454650/BambuBugReport.3mf.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12454669/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

superfloh247, 2023-09-06:
Hi, so the workaround is to minimally adjust the drawing height until the sides and the rounded corners get the same color after slicing?

zhimin-zeng-bambulab, 2023-10-11:
Yes, other than that, there is currently no good way to deal with this problem.