bambulab/BambuStudio #2419: List of supported G an M-Codes

markus512, 2023-08-29:
**Is your feature request related to a problem? Please describe.**
This is my only printer which does neither have any list of supported g-codes nor follows an open-source implementations g-codes. Allthough Marlin does fit for some of them there are many G and M-Codes which are undocumented. It's very annoying to work with such device.

**Describe the solution you'd like**
A clear and concise description of all G and M-Codes supported by Bambu-Lab printer.

**Describe alternatives you've considered**
Dumping firmware from a printer and reverse engineer this info which is a pain in the ass and definitely not my preferred way.
I also don't think It's Bambus preferred way as it would show me technical details which I probably should not see.

SaltWei, 2023-09-01:
@sodachen22 Should be handled by firmware team

markus512, 2023-09-01:
I'd just like to add that everything you can provide will be helpful to me. 
No matter whether it's a beautiful translated list or just a note written in chinese.

markus512, 2023-09-23:
I think I'll trash my printer and replace it with a better one (one which has good documentation or is open source)

closed 2023-09-23 by markus512

GuzziRaz, 2023-09-30:
@sodachen22 please reopen this issue. Redacting  gcode specifications does no good for anyone