bambulab/BambuStudio #2420: Organic supports from PrusaSlicer (vs Tree supports in BambuStudio)

Gnof, 2023-08-29:
**Is your feature request related to a problem? Please describe.**
The current tree supports aren't as functional as the organic supports that are currently in Prusa slicer.

Specifically, tree supports often overlap with the model and cause scarring / artifacts on the printed model surface:
[image]

**Describe the solution you'd like**
This issue does NOT occur on Prusa slicer with the same model with their implementation of tree supports (known as OrganicSupports in their source I believe? https://github.com/prusa3d/PrusaSlicer/blob/c05542590d7c2d73eb69bbf7a82a482a075815c1/src/libslic3r/Support/OrganicSupport.cpp)

Is it possible to port their supports implementation over to Bambu Studio?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

The following settings have been tested to attempt to remediate the issue: 
(Tested with all tree Styles [Default, hybrid, strong, slim])
- "On build plate only" - turned ON (off puts the supports directly on the model, causing more issues)
- "Support critical regions only" - tested ON and OFF
- "Remove small overhangs" - tested ON and OFF
- "Top Z distance" - set to 0, as well as adjusted +
- "Support / object xy distance" - set to larger numbers (2-3 mm+ to test) this actually results in the following: 
[image]

**Additional context**
Add any other context or screenshots about the feature request here.

I'd actually be happy to try and do this myself but I figured I'd put in a feature request in case it's either already on the roadmap or you guys have explored this and there's some reason it can't be done so simply (which would save me the time of trying to port the feature and finding out there's a deeper issue with it)

Thanks guys!

Gnof, 2023-08-29:
Sharing this comparison of the PrusaSlicer Organic support on the same model
[image]

Edzi11, 2023-08-29:
Would be nice to get an update on the support generation since they arent super good sometimes. I've had instances of support generating mid-air or in areas that had no need for support.

Noisyfox, 2023-08-31:
https://github.com/SoftFever/OrcaSlicer/pull/1938
Maybe try the latest Action build here.

Edzi11, 2023-08-31:
> [SoftFever/OrcaSlicer#1938](https://github.com/SoftFever/OrcaSlicer/pull/1938) Maybe try the latest Action build here.

Thanks will give this a try!

closed 2024-01-16 by tangketan