bambulab/BambuStudio #2426: Ability to add Printer by IP address (Lan Only Mode)

fire1ce, 2023-08-30:
Lan only mode uses auto-discovery to find a printer on local network. 
When using remote connection to the lan segment - like VPN, auto-discovery not always works and there is no way to use the printer (Communication to the mqtt works fine).

Add the ability to add printer manually hard coded by ip address / hostname (same as for klipper)

Alex-Schaefer, 2023-08-30:
Sadly, I don't think this is going to happen.
I don't know if this is deliberate to force their cloud, or if they are just incompetent, but I doubt Bambu will do anything.

There have been similar feature requests open for months.
#2232 the latest one
#1549 @StoneLiBambulab was assigned, but there was no comment on the problem.
#702 is months old and no Bambu response at all.

In #702 someone has linked a small script to send a fake SSDP message with the printer IP etc., to fool Bambu Studio and "discover" the printer, maybe this works for you.

fire1ce, 2023-08-31:
@Alex-Schaefer 
Thanks for the detailed response. I'll look at #702 if a workaround works, it's something.

Alex-Schaefer, 2023-08-31:
For me, it did the trick, and I have successfully tested it today.

But the original script is for Linux only.
Since I run Windows and have Python already installed, I created a small Python script to do the same. Should work on all platforms.
https://gist.github.com/Alex-Schaefer/72a9e2491a42da2ef99fb87601955cc3

I'm very glad that this is working (although it's a shame that something like that is needed), otherwise we probably would have returned the printer.

slynn1324, 2023-09-01:
2nd (or 5th?) this feature request.  Would be so much more reliable for those of us with more advanced network setups - VPNs or VLANs.  Just let me enter an IP/domain for the printer instead of forcing broadcast discovery.

QingZhangBambu, 2023-09-05:
OK, i will add it to our feature pool

ellenhp, 2023-09-13:
I got tired of waiting and reimplemented the entire proprietary networking plugin to add this myself. Leaving this here in case it helps someone.

https://github.com/ellenhp/bambu-farm

Znubbis, 2023-09-20:
+1 from me, we can't use LAN mode as the printer is on a wifi subnet.

fire1ce, 2023-09-29:
@walterwongbbl any progress on this ?

walterwongbbl, 2023-10-12:
> @walterwongbbl any progress on this ?
Hi.
We are considering feasible implementation methods

Alex-Schaefer, 2024-01-02:
Hi @walterwongbbl any estimated time on when this will be implemented?

walterwongbbl, 2024-01-22:
Hi @Alex-Schaefer 
非常抱歉，我们需要做的事情太多了。 我们将会重视这个问题。

Alex-Schaefer, 2024-01-24:
@walterwongbbl it's difficult to believe you are taking this seriously, if you can't even provide an estimated time for such a basic feature after 14 months.

hoegge, 2024-02-09:
It is quite crazy you cannot add a printer by IP - it means a lot of user simply cannot connect to it and means you can return the printer and get a refund, since it does not work.

moe5k, 2024-02-14:
+1 with the same problem, different subnets for WiFi and LAN and the WiFi Subnet not even have a connection to the internet. We really need to talk to the printer direct.

hoegge, 2024-02-25:
Mine showed up after selecting the "LAN only" mode on the printer. Have you tried that?

ellenhp, 2024-02-25:
> Mine showed up after selecting the "LAN only" mode on the printer. Have you tried that?

That's normal behavior for a router that doesn't block SSDP and situations where both machines are on the same subnet. There are unfortunately a couple corner cases where autodiscovery doesn't work which is what this issue is tracking. :)

ColColonCleaner, 2024-04-24:
@walterwongbbl Any news on this feature? I cannot use my X1C remotely because of this.

Kablahz, 2024-05-23:
@walterwongbbl Any traction on this issue. X1E was released for Enterprise environments. Addressing the printer without relying on SSDP is a must if marketing the product as Enterprise ready. Our purchasing of multiple X1E business wide is on hold until this issue is addressed.

timothyzalusky, 2024-06-11:
I really would like Bambu Lab to come out and tell us why they have not added this feature.  This is essential for business use of these printers.  It does not make sense they have not added it yet.

fidesachates, 2024-06-12:
Want this to happen as well. +1

ColColonCleaner, 2024-07-30:
A collaborator mentioned in #2232 that this will be added in the next update. I hope that actually happens, and am looking forward to it if it does.