bambulab/BambuStudio #2429: Linux AppImage fails to run without installing libwebkit2gtk-4.0-37

officiallymarky, 2023-08-30:
### Bambu Studio Version

1.07.04.52

### Where is the application from?

Bambu Lab github releases

### OS version

kubuntu 23

### Printer

X1C

### How to reproduce

Download App Image
chmod +x
Run

### Actual results

"The Command Could not be found error"
If you run it from command line, you can see it is missing shared library libwebkit2gtk-4.0-37.

### Expected results

It should load.

### Project file & Debug log uploads

sudo apt install libwebkit2gtk-4.0-37 solves it, but this should be included with the appimage so there is no dependencies.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Dapid, 2023-12-19:
Same on Fedora. The name of the package it requires is `webkit2gtk4.0`

Jubei-VA, 2024-01-17:
Same issue here, Ubuntu 23.10

jeroenweij, 2024-05-03:
libwebkit2gtk-4.0-37 is not available on ubuntu 24.04

lanewei120, 2024-05-06:
similiar as https://github.com/bambulab/BambuStudio/issues/3973

We will try to include it into appimage
the problem is we are not familiar with linux, and don't know how to solve it currently

MackBambu, 2024-06-27:
https://github.com/MackBambu/BambuStudio/actions/runs/9691686545/artifacts/1643561459
This appimage was built using libwebkit2gtk-4.1.
Can you help me test if it works?@jeroenweij

jeroenweij, 2024-06-27:
> https://github.com/MackBambu/BambuStudio/actions/runs/9691686545/artifacts/1643561459 This appimage was built using libwebkit2gtk-4.1. Can you help me test if it works?@jeroenweij

Yes I can!
It seems to work. I did not stat any print but the program loads and works normally, before it would not startup at all.

CaptainCoward, 2024-12-14:
This is still a thing on the latest builds.