bambulab/BambuStudio #2430: Multicolor printing AMS Slot 1 is not considered

Guenni1975, 2023-08-30:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1C

### How to reproduce

I have a multicolor print.
Slot 1 Black
Slot 2 White
Slot 3 Cyan

Model was colored manually.

First picture preview how it looks colored!
The black from slot 1 is not taken into account after slicing.
If I take the red from slot 4 it will be considered.

Preview
[image]

Color Slot 1
[image]

Color Slot 2
[image]

Apparently, the color in AMS slot 1 is currently not taken into account for multicolor printing!
The colors in other slots are considered!

### Actual results

The color in AMS Slot 1 is not taken into account in multicolor printing!

### Expected results

The color in AMS slot 1 would have to be taken into account!

### Project file & Debug log uploads

[beispiel.zip](https://github.com/bambulab/BambuStudio/files/12478621/beispiel.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Guenni1975, 2023-08-30:
I have checked it again, with another file the coloring in 3 colors also works with slot 1 without problems. So it seems to be related to the file. I just don't understand why it refuses to colorize only slot 1 with the file, but works with slot 4. In so far, the bug or what it is not so bad for now!

Guenni1975, 2023-09-06:
The problem also occurs with four-color printing.  However, it cannot be bypassed there because all slots are used.  So it should be fixed soon.

zhimin-zeng-bambulab, 2023-10-24:
we will fix it in 1.8 beta

zhimin-zeng-bambulab, 2023-11-13:
It has been fixed in v01.08.00.57, you can have a try.

closed 2023-11-13 by zhimin-zeng-bambulab