bambulab/BambuStudio #2432: Travel acceleration option request

kr320, 2023-08-31:
I would like to have an option for choosing the travel acceleration independent of printing acceleration. I'm sure you're aware this is available in Orca slicer so I'm curious as to know why it isn't in Bambu Studio.

I generally like to slow prints down since most of the time I'm not in a rush or want better quality. Having the ability to decrease Normal printing acceleration and increase travel acceleration would be a nice option to have.

mzient, 2023-10-18:
Bumping up! The slow travel acceleration is especially visible in the 1st layer.

kr320, 2023-10-22:
> Bumping up! The slow travel acceleration is especially visible in the 1st layer.
> 
> 

For sure. I'm surprised this still hasn't been addressed or implemented yet. It's a feature in orca slicer already though I don't believe it changes first layer travel accel.

I vaguely remember seeing in update notes something along the lines of them changing acceleration for maybe first layer, I think it had something to do with the pre print prep stuff. It's probably modifiable in machine gcode somewhere. Im not super great at that stuff. 

If you do look into it and find something out please do let me know! Increasing first layer travel speed even a little bit would be nice.