bambulab/BambuStudio #2435: System Presets

mcdavidr, 2023-08-31:
### Bambu Studio Version

1.7.452

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu P1S

### How to reproduce

[image]
There as no presets for P1S, only X1C

### Actual results

Only presets for X1C

### Expected results

P1S presets?

### Project file & Debug log uploads

[example.zip](https://github.com/bambulab/BambuStudio/files/12491618/example.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2023-09-14:
This preset is also applicable to P1S, so we did not create a separate  preset and name it xxx@P1S.

closed 2023-09-17 by mcdavidr

weczi, 2024-09-09:
> This preset is also applicable to P1S, so we did not create a separate preset and name it xxx@P1S.

Thats preset dont works for me. every first layer ruined.