bambulab/BambuStudio #2440: Fix the filament selector

beehphy, 2023-09-02:
[image]
 Why would you select another filament after I have specifically selected one? You waste my time. waste my material. WTF? It is so frustrating to return to a half-print to find that the very last step of sending the job was ruined by the idiot slicer choosing the wrong material just because the same COLOR was used in the picker. Silk, sparkles, translucent, fiber reinforced. there are so many reasons you might have similar colored materials, but NOT want to JUST INTERCHANGFE THEM! don't decide something for me!

beehphy, 2023-09-02:
Why do the colors in the slicer use numbers only? 
And not what the machine uses; alpha and numeric?
This is where you are creating the disconnect.

QingZhangBambu, 2023-09-05:
@beehphy If there are cases of the same color, how about materials with the same ID in the user slicer have priority?

beehphy, 2023-09-05:
that seems like the proper behavior: 
Do not change away from user selected filaments by default.

bam-winterfell, 2023-12-21:
@beehphy We have made some modifications in the new version. Please check if the issue has been resolved.

closed 2024-01-18 by bam-winterfell