bambulab/BambuStudio #2444: Screen goes white after slice

mroberts1408, 2023-09-03:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz   3.60 GHz Windows 10 22H2 Home

### Printer

Bambu P1S

### How to reproduce

Import file
click on slice plate

### Actual results

Once you slice, the screen goes white
[image]

### Expected results

It should show a preview

### Project file & Debug log uploads

[debug_Sun_Sep_03_11_37_21_17380.log.zip](https://github.com/bambulab/BambuStudio/files/12506494/debug_Sun_Sep_03_11_37_21_17380.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

QingZhangBambu, 2023-09-15:
could you provide your 3mf file, i tend to reproduce it.

mroberts1408, 2023-09-15:
I got it figured out, graphics card hadn’t updated for some reason. I
updated the card and all the issues went away.

QingZhangBambu, 2023-09-18:
@mroberts1408 OK. I‘m glad to know that.

closed 2023-09-18 by QingZhangBambu