bambulab/BambuStudio #2445: Ability to skip "checking support necessity" step of slicing when supports are disabled

ctrl-alt-breathe, 2023-09-03:
Having to wait for minutes on complicated prints for the step called "Checking support necessity" when supports are explicitly disabled is very annoying and adds a lot of time to the already sometimes long slicing process.
It would be very helpful to have something like a skip button during slicing or an option under support settings to disable this step.

tangketan, 2023-09-04:
I'm also thinking of opening an option to disable it, but I'm afraid of too many options. How about re-using the option "slow down for overhangs"? If  "slow down for overhangs" is turned off, it means the users are certain the model has no overhangs, then there is no need to check support necessity.

SaltWei, 2023-09-04:
> slow down for overhangs

No. They are different settings

ctrl-alt-breathe, 2023-09-04:
Yeah, they are different. Even if it was something thrown in settings somewhere so it's not cluttering the print settings, it would be a huge help.

kemper, 2024-08-08:
I would also love this feature. Some models this stage take abnormally long and I'm just playing around with infill and layer patterns and such.

JansenSmith, 2024-12-05:
This basically adds hours to my workflow for no reason, given that I am explicitly disabling supports.