bambulab/BambuStudio #2446: K values don’t save on Italia Language

djmarkino94, 2023-09-04:
### Bambu Studio Version

01.07.04.52

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS Ventura 13.5.1

### Printer

Bambulab x1c combo

### How to reproduce

Insert k values on manage and it doesn't save them

### Actual results

The k value is not stored On italian language

### Expected results

the values ​​should be memorized as when bambu is in English

### Project file & Debug log uploads

Q

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-09-05:
@djmarkino94  Could you share your log file when you can't save it?

djmarkino94, 2023-09-05:
how should i do?Inviato da iPhone

djmarkino94, 2023-09-05:
> @djmarkino94  Could you share your log file when you can't save it?

how should i do?

zhimin-zeng-bambulab, 2023-09-05:
[image]
@djmarkino94  First find the log folder, then find the log file corresponding to your current operation time.

djmarkino94, 2023-09-05:
Ok but how do I know which is the right log file to send? how do i know that log shows the error?

zhimin-zeng-bambulab, 2023-09-05:
1. You can delete all the original logs before starting the software, or backup them to other folders, so that the logs generated after you start the software are what you want.
2. After starting the software, reproduce this bug in the Italian environment, and then upload the newly generated log file to github. Let me take a look at the log file.

djmarkino94, 2023-09-05:
let me you know if its ok

zhimin-zeng-bambulab, 2023-09-05:
I found that when you saved the data, there is already a problem. Can you describe the steps to reproduce this problem in detail?

djmarkino94, 2023-09-05:
very simply when it is in English I can see the stored K values, when I switch to Italian the K values ​​are all zero and when I change them they return to zero, in practice it does not save them. I try to attach a video

djmarkino94, 2023-09-05:
[Senza nome.mov.zip](https://github.com/bambulab/BambuStudio/files/12522639/Senza.nome.mov.zip)

djmarkino94, 2023-09-08:
> I found that when you saved the data, there is already a problem. Can you describe the steps to reproduce this problem in detail? And what is the diameter of the nozzle you are using？

Hello News?

sodachen22, 2023-09-09:
Yes, we know the reason. The printer only support ".", but in Italian, "." is instead of ",". We will fix this issue. Please use English as a workaround before the new software with this fix.

djmarkino94, 2023-09-09:
> Yes, we know the reason. The printer only support ".", but in Italian, "." is instead of ",". We will fix this issue. Please use English as a workaround before the new software with this fix.

Thank you

Witawat, 2023-09-17:
i have problems in English Language

[image]

[log.zip](https://github.com/bambulab/BambuStudio/files/12643068/log.zip)

djmarkino94, 2023-10-10:
- [ ] News?  I always have the same problem

zhimin-zeng-bambulab, 2023-10-10:
@djmarkino94  Please provide the Studio version and firmware version in which you still have this problem. Thanks!

djmarkino94, 2023-10-10:
> @djmarkino94 Please provide the Studio version and firmware version in which you still have this problem. Thanks!

Same problem, on Mac when I put the Italian Language the K value its all to 0, and if I try to modify dont'change.
[image]
[image]
[image]

zhimin-zeng-bambulab, 2023-10-13:
maybe the same as #2689
@djmarkino94  It has been fixed in V1.7.7.89, you can have a try.

closed 2023-10-13 by zhimin-zeng-bambulab