bambulab/BambuStudio #2447: Nozzle purge position not in center of container

redsector72, 2023-09-04:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Macos 11.7.9

### Printer

Bambulab X1C

### How to reproduce

Simply print anything after slicing

### Actual results

The purge/noozle cleaning does not purge as deep into the poop shoot as before.
It seems to poop just on the edge, and as such the poop sticke to the side of the container and sometime this causes to drags poop onto the plate.

### Expected results

Poop goes down the chute

### Project file & Debug log uploads

[Cube.3mf.zip](https://github.com/bambulab/BambuStudio/files/12512816/Cube.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pichen2021, 2023-09-05:
Hi
    The purge position is a fixed coordinate, maybe it's because a slight change in the printer, could you please:
1. Retension the belt (https://wiki.bambulab.com/en/x1/maintenance/belt-tension);
2. Gently push the tool head over the excess chute and to the limit along the Y axis, check the position of the nozzle in the excess chute, and check if the metal is installed correctly in the chute as below, and please refer to this [wiki](https://wiki.bambulab.com/en/p1/maintenance/excess-chute) to check if the excess chute is loose
[image]
3. If it is still abnormal, please take a video of the tool head moving to the excess chute and purging.

Thanks.

redsector72, 2023-09-12:
Seems regular according to this guide, but still the filament get stuck on the bin. I suggest adding to use PTFE oil in the chute in the 30 days maintenance

renewedbullet, 2024-01-07:
change the purge location to center of shoot to help prevent build up or hitting the wiper to uch and causing damage