bambulab/BambuStudio #2448: 校准：温度塔，温度不正确

dflstone, 2023-09-04:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

voron2.4

### How to reproduce

1.准备->校准->温度->PLA|start temp:230|end temp:190->确认->切片单盘->导出G-Code文件
2.打开导出的Gcode文件->搜索M104->除了初始温度，打印过程中M104 S190开始递减5度，最后M104 S150

### Actual results

逻辑有问题，230-190 变成了 190-150度

### Expected results

逻辑有问题，230-190 变成了 190-150度

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dflstone, 2023-09-04:
切片后再界面颜色方案中选择温度也是从190-150度显示不同温度

zhimin-zeng-bambulab, 2023-10-08:
已经在 v1.7.6 中修复

closed 2023-10-08 by zhimin-zeng-bambulab