bambulab/BambuStudio #2449: Object Printing Project issues

Benderrodrigu3z, 2023-09-04:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 10.15.7

### Printer

X1 Carbon

### How to reproduce

In the slicer change printing sequence to object:

pressing a number does not change filament type(in Print by object mode), also 0 should make it selection 10

When using multiple plates, they cant act as there own for a project, if I want to make 2 of them as print by object and the others as print by layer I have to change it every time before a print, these should be moved to a right click menu that says for this plate do this and ignore the global, strength, quality, these are only overrides though, so you don't need things like prime tower

### Actual results

it doesn't work

### Expected results

said above

### Project file & Debug log uploads

any project with multiple plates

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-09-05:
this is 3d area operation

Haidiye00, 2023-09-05:
Can you record a video or take a screenshot? Thank you

Haidiye00, 2024-05-14:
Please download and use the latest version. If there are still any issues, please use pictures or screenshots to describe your ideas. Thank you

closed 2024-05-14 by Haidiye00