bambulab/BambuStudio #2451: create an AMS 2.0 variant that can tolerate flexible filaments

malefactor3952, 2023-09-04:
**Is your feature request related to a problem? Please describe.**
Create an AMS 2.0 variant that can tolerate flexible filaments

**Describe the solution you'd like**
AMS works well.  its only limitation seems to be flexibles.  multi-color flexibles would be great, and I'd buy 1-2 new AMS immediately.

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

sodachen22, 2023-09-05:
Thanks for the supporting. Yes, we are also investigating the solution to make flexible multi-color flexible filament work with the printer. There are some technical issues that need to be resolved. Can you please also share with us if you have ideas for this?

Noisyfox, 2023-09-05:
Yes please. I'd like to embed TPU connections in normal materials which is not possible with current AMS.

malefactor3952, 2023-09-05:
I have seen some research/studies that seem to suggest that with TPU, you can't rely on the extruder to pull on the TPU to unspool it -- or defects will show up in the print.  It's hard to push through a bowden tube.  The main printer's extruder must be the only one in primary control of tension/load/unload.  This direct-drive-extruder must be able to exert minimal force and feel minimum impedance in both directions.  

You currently "stage" filament in a bowden tube to be ready to print, knowing all you have to do is pull it.  But with TPU you almost need to find a way to stage 10cm of filament just drooping, sitting somewhere (near the printer?) so that it's ready to be printed with, no resistance in extract or retract.  The problem with this approach is that it totally breaks with rigid filaments; they would not droop.  So you have to know a priori that you are working with a flexible filament.

You could maybe autodetect rigidity of a filament by trying to extrude + retract-at-AMS at the same time, and use stepper motor feedback.  Not sure if this can be done in a way that doesnt eventually (over time, at scale) lead to a problem (broken filament in extruder, requires disassembly, not user-friendly).  If you could do this, the information could be fed in and auto-adjust to guarantee a good print.  TPU is notoriously inconsistent, and the 'hardness' ratings provided by manufacturer don't mean much.  

Maybe such a droop autodetector could be assembled of (relatively) inexpensive parts and in the shape of the multi-AMS-hub (but larger) on the back of the X1C.

malefactor3952, 2023-09-05:
++add to above.  AMS would need to assume responsibility of making sure sufficient filament was maintained as 'staged'.  So in a sense the 2 steppers act asynchronously, each having their own job (main extruder prints, AMS extruder maintains sufficient slack ready-to-print).

sodachen22, 2023-09-09:
Thanks, @malefactor3952

kylek29, 2023-09-12:
> ++add to above. AMS would need to assume responsibility of making sure sufficient filament was maintained as 'staged'. So in a sense the 2 steppers act asynchronously, each having their own job (main extruder prints, AMS extruder maintains sufficient slack ready-to-print).
> 
> Also, all of the above was written with the thought of "how do you solve for printing with the softest, most flexible filaments?". I was thinking about the 70 shore hardness stuff, that is really difficult to deal with.

I believe that's how the Chameleon3d went about working with flexibles.