bambulab/BambuStudio #2452: Im Höhenbereich Modifikator in Deutsch fehlt das + Zeichen

Zamaris39, 2023-09-04:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

BambuLab P1P

### How to reproduce

1. Rechtsklick auf das Modell
2. Höhenbereich einstellen
3. + fehlt
[image]

### Actual results

Es ist nur 1 Höhenbereich einstellbar.

### Expected results

Es sollen mehrere Höhenbereiche einstellbar sein.

### Project file & Debug log uploads

Das gilt für alle Modelle mit mehrfarbigen Höhenbereichen.
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-09-05:
[image]
you can click each height rang tool to set the range you wanted

closed 2023-10-24 by QingZhangBambu

UVM24, 2023-11-24:
Habe das selbe Problem. Hast du eine Lösung gefunden? Pro Objekt sollten mehrere Einstellungen möglich sein. Im Web habe ich Bilder gefunden, in welchen das mehrfache Modifikation abbildet. Allerdings nur auf amerikanischen Seiten.
[image]