bambulab/BambuStudio #2453: Camera stopped working after selecting 1080p, it won't go back to 720p and says no camera.

alrididan, 2023-09-04:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu lab x1 Carbon

### How to reproduce

Device and click on 1080p for the camera.

### Actual results

the camera worked for a small period of time and then stopt working at all,
now i can't go back tot device and click on 720p because it is not selectable anymore?

### Expected results

Bambu lab x1c saying there is no camera device and  no spaggetti detection

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12515537/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-09-05:
Can you change video size from Device Panel of Bambu Studio?
[image]

alrididan, 2023-09-05:
No I can’t select that, I see the option to make the 720 dot green but nothing happens when I click on it an it stays on 1080 green dot.

Verstuurd vanaf mijn iPhone

[image]<https://user-images.githubusercontent.com/112537880/265563173-38f11724-79fa-407a-bab0-1fca127656a7.png>

alrididan, 2023-09-05:
Just got home and did what you say, it jumps to the green dot on 720p but still says no camera. I had a look inside where the connector from the camera is and there is a flashing red light on that board.

[image0.jpeg]
[image1.png]Verstuurd vanaf mijn iPhone

Then can you change video size from the printer screen?
[img_v2_b3622be5-3503-4786-92de-0b95b392476g]<https://user-images.githubusercontent.com/112537880/265602212-91a9165d-5dd1-4cc3-ada6-5634b94baad8.jpg>

alrididan, 2023-09-06:
Also did a factory reset today, but that too still gives the same error as if there is no camera. because I wanted to print anyway, I turned on a file from the sd card and let it print. when I was curious about how long it would take, I turned on bambustudio to see how long that would take. I then clicked on the camera and to my surprise it worked!! unfortunately this was short-lived because 5 minutes later the same error message came back and I have the message that I do not have a camera again, this is both on bambustudio and on the printer itself. If I may make my own assessment, it seems that the printed circuit board to which the camera is connected is defective or still a software error.

bambu123, 2023-09-07:
> Also did a factory reset today, but that too still gives the same error as if there is no camera. because I wanted to print anyway, I turned on a file from the sd card and let it print. when I was [...]

@sodachen22 It seems to be a physical connection problem with the camera.

sodachen22, 2023-09-07:
@alrididan  Could you please reseat the camera and test again according to this guide https://wiki.bambulab.com/en/x1/maintenance/replace-chamber-camera? If the issue persisted, please create a ticket and upload the log (https://wiki.bambulab.com/en/x1/troubleshooting/upload-log), there may need a replacement of the camera.

alrididan, 2023-09-07:
I have had the camera loose, at first sight the problem does not seem to be in the plug! perhaps it would have been useful if I had mentioned this earlier, but the WiFi connection also sometimes fails. the first few minutes it usually goes well, but then suddenly he loses the connection with bambu studio.

Verstuurd vanaf mijn iPhone

sodachen22, 2023-09-07:
@alrididan Could you please also check the antenna of the WiFi module (https://wiki.bambulab.com/en/x1/maintenance/replace-wifi-antenna)? And please alse make the printer to be closer to the router or create a hotspot with mobile phone and make the priner connect to the hotspot to have a double confirm.
If there is problem with hardware, please create a ticket together with LOG and some pictures or videos

alrididan, 2023-09-11:
Sorry for my late response, I don't really know what to do now. Last Friday I turned the printer back on and everything worked as if nothing was wrong. To sum it up: WiFi no problem and the camera worked just fine? as if this email and complaint were unnecessary. So I'm now waiting to see whether I should submit a ticket and what exactly is going on, but since I don't print every day and this is only a hobby for me, it might take a few weeks before I know whether it is still necessary and maybe an update will solve it completely and I will never have to submit a ticket. let's hope so!!

alrididan, 2023-11-20:
Hello,
the WiFi connection now seems stable after replacing the router and moving the printer closer. the WiFi connection also remains, but now the printer says that there is no camera when it is switched on? I checked the wiring and disconnected and reconnected it, but that doesn't seem to solve anything. In summary: there is a WiFi connection but no camera, as if there is no camera.

What else can I do?

Yours sincerely,

Edward Meijer

walterwongbbl, 2023-12-21:
Hello, we are very sorry to inform you that we are busy with the release of a new version. May I ask if your issue has been resolved

closed 2024-01-18 by walterwongbbl