bambulab/BambuStudio #2455: Cannot save project file with certain characters

kvnper, 2023-09-05:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

X1C

### How to reproduce

1. save project
2. use the following filename:  p1_PLA_x1c-0.4n_BS1.7

### Actual results

Error message comes up "Failed to save the project. Please check whether the folder exists online or if other programs have the project file open."

### Expected results

Project saves successfully

### Project file & Debug log uploads

Not necessary

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-09-06:
the name is valid. maybe there are other problems.

kvnper, 2023-09-06:
I did not get this error in previous versions, I always use this naming scheme

kvnper, 2023-09-06:
Log file attached here, apologies for not including it earlier.
[log.zip](https://github.com/bambulab/BambuStudio/files/12533832/log.zip)

kvnper, 2023-09-07:
I just discovered that if i remove the ".7" from the end of the filename then it works. It must be the period/full stop that messes it up.

bambu123, 2023-09-11:
> I just discovered that if i remove the ".7" from the end of the filename then it works. It must be the period/full stop that messes it up.

What do you mean by the 'period/full stop', is there a typo error when input the project name for save? Do you still think this is a bug?

kvnper, 2023-09-11:
> > I just discovered that if i remove the ".7" from the end of the filename then it works. It must be the period/full stop that messes it up.
> 
> What do you mean by the 'period/full stop', is there a typo error when input the project name for save? Do you still think this is a bug?

Let me clarify:

"p1_PLA_x1c-0.4n_BS1" works and "p1_PLA_x1c-0.4n_BS1.7" does not work.

Yes, I think it is still a bug.

bambu123, 2023-09-11:
> Let me clarify:
>
> "p1_PLA_x1c-0.4n_BS1" works and "p1_PLA_x1c-0.4n_BS1.7" does not work.
>
> Yes, I think it is still a bug.

Thanks. I found that you save files in OneDrive. If you save them in a normal directory, will there be the same problem?

kvnper, 2023-09-11:
No, does not work. Same problem.

bambu123, 2023-09-11:
> No, does not work. Same problem.
> […](#)

Please send me new logs containing normal and error situations.  That is "p1_PLA_x1c-0.4n_BS1" vs "p1_PLA_x1c-0.4n_BS1.7"

kvnper, 2023-09-11:
Attached 
[Uploading log 230911 2007.zip…]()

kvnper, 2023-09-11:
Attached
[log 230911 2007.zip](https://github.com/bambulab/BambuStudio/files/12574781/log.230911.2007.zip)

bambu123, 2023-09-11:
You were saving "p1_PLA_x1c-0.4n_BS1" at 2023-09-11 20:07:14. It that true? Did you save "p1_PLA_x1c-0.4n_BS1.7" before or after that?
But I can't find the log for saving with name "p1_PLA_x1c-0.4n_BS1.7", it may be caused by the bug not to save.

kvnper, 2023-09-11:
Yes I tried before or after, I can't remember which.

bambu123, 2023-09-12:
> Yes I tried before or after, I can't remember which.

Can you help test it with the internal version below? I have added some related logs.
https://1drv.ms/u/s!AkMfUF_oU9pBniPN-qsufR6DL7xS?e=uObZFw

kvnper, 2023-09-12:
Same problem as before
[image]

kvnper, 2023-09-12:
Same problem as before
[log 230912 1619.zip](https://github.com/bambulab/BambuStudio/files/12583022/log.230912.1619.zip)
[image]

bambu123, 2023-09-12:
> Same problem as before [image]

Is the type list of your file save dialog box different from mine?
[image]

kvnper, 2023-09-12:
Yes 
[image]

bambu123, 2023-09-12:
> Yes [image]

Your file save dialog box does not automatically add the extension. If you enter test instead of test.3mf, the save may also fail.

kvnper, 2023-09-12:
I just tied, it worked with just the following name: 
[image]

cmguo, 2023-09-12:
> file save dialog box does not automatically add the extension.

I can reproduce the problem here with 'xxxx.7z'.

bambu123, 2023-09-12:
Thanks for your help. We found the reason. ".7" is the registered file type on your system. The file saving dialog box will not automatically add the ".3mf" suffix.
This may be a Windows system issue, as discussed here.
[stackoverflow.com/questions/55711039/savefiledialog-addextension-doesnt-work-as-expected](https://stackoverflow.com/questions/55711039/savefiledialog-addextension-doesnt-work-as-expected)

kvnper, 2023-09-12:
Got it, no problem. Not a big issue for me, I can work around it.

Thank you.

bambu123, 2023-12-21:
Fixed in v1.8.2.

closed 2023-12-21 by bambu123