bambulab/BambuStudio #2456: Print head has a very small pause right after wipe while retracting movement.

npoulos, 2023-09-05:
### Bambu Studio Version

1.6.6

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Printer

Slicer specific

### How to reproduce

Slice a project with "Wipe while retract" turned on

### Actual results

When you use "Wipe while retracting" it moves the head across the wipe movement while retracting but then has a small pause where it retracts a tiny bit while stationary.

This tiny pause is causing stringing when printing tiny models with a .2 nozzle.

[image]

### Expected results

The retraction should complete with the last wipe move, or even before the wipe move finishes to ensure no oozing.

Please update to allow for either
- the complete retraction to happen during the wipe movement
- control over how far into the wipe movement that the retraction finishes

For example, you could have the wipe be at 1mm and retract at 0.8mm, and a new setting to complement "wipe while retracting" that is called "Retract over portion of wipe movement" and it can be a %. 100% would be that the wipe finishes right at the end of the wipe line and 50% would be that the retract happens over the first 50% of the wipe move and then last 50% of the wipe move will have no retraction.

### Project file & Debug log uploads

[BBS-retract.3mf.zip](https://github.com/bambulab/BambuStudio/files/12522899/BBS-retract.3mf.zip)
[BBS-retract.3mf.zip](https://github.com/bambulab/BambuStudio/files/12522906/BBS-retract.3mf.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

QingZhangBambu, 2023-10-25:
this is to ensure the entire retraction length is honored in case the wipe path is too short

npoulos, 2023-10-25:
@QingZhangBambu in that case it should have logic to first check if the wipe path is too short or not and then add the last bit. Otherwise if the path is long enough when it does this last little bit of retraction it pauses the print head momentarily which causes tiny remnants.

QingZhangBambu, 2023-10-26:
I agree with you. So I will check this code.