bambulab/BambuStudio #2457: AMS settings button doesn't show clearly in Japanese, macOS.

lilyharuna, 2023-09-05:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab github releases

### OS version

macOS Sonoma 14.0

### Printer

Bambu Lab P1P + AMS

### How to reproduce

1. Install Bambu Studio in macOS.
2. Go to settings(command + ,) turn language to Japanese.
3. Connect AMS in P1P.
4. In the device tab, the 'Retry(再試行)' button will hidden and doesn't show clearly.

### Actual results

The right half of the retry(再試行) button is hidden and cannot be seen. (Also, the right half of the hint is hidden.)

### Expected results

All button is clearly shown.

### Project file & Debug log uploads

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

walterwongbbl, 2023-09-11:
Hi. @lilyharuna 
we will optimize it

walterwongbbl, 2023-10-13:
Hi. @lilyharuna 
Please download and use the latest Studio version

walterwongbbl, 2023-12-21:
If you have any other issues, please create a new issue. The current conversation will be closed.

closed 2023-12-21 by walterwongbbl