bambulab/BambuStudio #2461: Home Screen Displays Blank, No Login Button

bhameline, 2023-09-05:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: AMG Ryzen 7950X
RAM: 64 GB
GPU: NVIDIA RTX A2000

### Printer

Bambu Labs X1C

### How to reproduce

Open software. Blank screen with navigation ribbon.
Click on 'HOME' icon, blank screen with navigation ribbon.

### Actual results

The app shows a blank 'HOME' screen. No 'Login' button.

### Expected results

There should be a 'Login' button and a 'Recents' button and User Manual button.
This is screen performs correctly on a different laptop on Windows 11.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12528489/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

tonycstech, 2024-03-16:
Software comes with a dependency on a Microsoft Edge also known as a VIRUS.
Most smart people disable and remove VIRUSES but bambu lab decided to include it and make it part of the software.
Thanks ! 4 out of 5 stars printer review removed

theboyknowsclass, 2024-04-01:
I have this same issue on 1.8.4.51 - bambu studio takes a long time to run and then doesn't load the home page so cannot log in.

would it be better to package your Webview in if that is causing the issue?

theboyknowsclass, 2024-04-01:
or if there is a fixed version I need let me know? (I'm using the evergreen webview2)