bambulab/BambuStudio #2466: Saved Flow Dynamics not available across multiple machines.

randyre, 2023-09-06:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C and P1S

### How to reproduce

1. Calibrate a filament on the X1C and save its values and name.
2. Go to P1S and attempt to pick that newly calibrated profile and it's not there, nor is the option to pick any profile.

### Actual results

[image]

### Expected results

All filament profiles should be available across all machines, regardless of where the profile was created.

### Project file & Debug log uploads

No file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

sodachen22, 2023-09-11:
The flow calibration results are related to the physical parameters of the printer, the result from one printer can not be used to another.

randyre, 2023-09-12:
@sodachen22 @QingZhangBambu 

sodachen22,

I not sure that statement is correct since the calibration is able to be run once on an X1 and it's then available to all X1's, I just want to be able to use a calibration from my X1 with one of the P-series printers.

Calibration is more about the characteristics of each individual filament than it is about the machine.  Now if we were comparing a Creality Cr-6 to the X1 I can see it being really different, but the numbers should be nearly identical between a X1, P1P, and P1S, and regardless they would be better than no calibration even if it off a little due to machine specifics.

randyre, 2023-09-12:
I'm wondering if it's related to not having an AMS attached to the P1S, since there is not even an option to save a profile.