bambulab/BambuStudio #2467: Calibration errors on Multicolor Filaments

randyre, 2023-09-06:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C and P1S

### How to reproduce

Filaments that have multiple colors within the same filament fail the automated calibration test.
[image]

### Actual results

Calibration Fails.

### Expected results

Calibration should complete

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2023-09-11:
@randyre 
There is a probability that the printer may fail to calibrate. In this case, you can make several attempts: The calibration line is uneven, 2. The laser radar is blocked. Restart calibration printing to see if the problem still exists.

randyre, 2023-09-12:
It fails every time on this multicolor filament and 90% of the time on silk filaments, I believe it's a color and reflectivity issue with the  LIDAR camera.

xenon2008, 2023-09-19:
i also got this error on a normal Black ABS... 
Cleand the lens, restartet the printer ....

But it is the same :(

zhimin-zeng-bambulab, 2023-10-11:
@randyre Has this issue been improved with the latest firmware version?

zhimin-zeng-bambulab, 2023-11-23:
Printer calibration success rates have improved, you can update the firmware version to v1.6.6.0 and verify it.

closed 2023-11-23 by zhimin-zeng-bambulab