bambulab/BambuStudio #2471: Many print stoppage alerts not sent to Bambu Handy

matmire-makes, 2023-09-06:
### Bambu Studio Version

handy

### Where is the application from?

Bambu Lab Official website

### OS version

iOS 16.6

### Additional system information

iphone

### Printer

X1C

### How to reproduce

Get an AMS retraction error or other notification.

### Actual results

The print stops, and an alert shows up on the printer screen, but no alert is sent to Bambu Handy... So, it sits for two hours before I notice that the print isn't going.  Wasted a ton of valuable print time.

### Expected results

printer should send an alert to my phone so that I can fix it and continue the print right away.  To be clear, I do get alert notifications on my phone sometimes, but it is very inconsistent.  

### Project file & Debug log uploads

you don't need a project file. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

TaylorN15, 2023-10-15:
Are there any updates on this? I have the same issue   I usually get a notification when a print is completed or cancelled, but I ever get any notifications about AMS issues etc. during the print

MrZepto, 2023-12-14:
Still no update on this? Same issue still.

SaltWei, 2024-01-16:
Not studio issue, close.
should be move to app https://github.com/bambulab/Bambu-Handy/issues

closed 2024-01-16 by SaltWei