bambulab/BambuStudio #2474: video  stopped working after 50%

BeachGeek92009, 2023-09-07:
### Bambu Studio Version

1.7.4.52

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 12.6.7

### Printer

Bambu Lab x1 c

### How to reproduce

1.start bambu studio 
2.submiited print job
3. video working
4. after about 50% video is blank ,nothing changed.
[image]

### Actual results

video is stil blank at 74%
[image]

### Expected results

Expect to see video
[project file.3mf.zip](https://github.com/bambulab/BambuStudio/files/12544011/project.file.3mf.zip)
[debug_Wed_Sep_06_19_18_19_64716.log.0.zip](https://github.com/bambulab/BambuStudio/files/12544023/debug_Wed_Sep_06_19_18_19_64716.log.0.zip)
[project file.3mf.zip](https://github.com/bambulab/BambuStudio/files/12544033/project.file.3mf.zip)

### Project file & Debug log uploads

[project file.3mf.zip](https://github.com/bambulab/BambuStudio/files/12544205/project.file.3mf.zip)
[debug_Wed_Sep_06_17_16_35_63669.log.0.zip](https://github.com/bambulab/BambuStudio/files/12544221/debug_Wed_Sep_06_17_16_35_63669.log.0.zip)
[debug_Wed_Sep_06_19_18_19_64716.log.0.zip](https://github.com/bambulab/BambuStudio/files/12544223/debug_Wed_Sep_06_19_18_19_64716.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-09-11:
We found in the log that after the playback connection was disconnected once, connecting the printer again has been failing, which may be caused by an abnormal network environment. Has the problem persisted in recent days?

BeachGeek92009, 2023-09-14:
I watched the Youtube videos on how to connect with Lan Only/Live view, since doing this camera is active 100% of the time. 
This issue can be closed. Thank you for looking into it.

bambu123, 2023-09-14:
OK, this question is closed first.

closed 2023-09-14 by bambu123